riverloopsec/killerbee #275: can use USRP B210for killerbee

ashok5141, 2023-06-26:
Hello,

I have created zigbee network using Raspberry pi 4B and Raspbee II module, I wish to exploit zigbee network, trying for university project, currently i have USRP B210 and HackRF One. 
can I able to killerbee using either of these tools.  suggest me some other tools to test the security of zigbee network using USRP B210 or HackRF One.

Looking forward your responses, Thanks in advance.
[image]

SullyM30, 2024-02-01:
Hi @ashok5141 I’m also a university student trying to do the same. Did you get it working?

ashok5141, 2024-02-02:
Hi @SullyM30 killerbee is not working, https://github.com/bastibl/gr-ieee802-15-4
I used block diagram only for sniffing of communication.