riverloopsec/killerbee #283: Where to buy an APIMote in 2024?

ep2277, 2024-04-02:
We are looking to buy an APIMote, but it seems to be almost impossible to get one.
Can you help?

tc-26, 2024-04-03:
I don't know of any available or anyone producing them. Is there something specific you're trying to do with the Apimote that other hardware won't support? I use the TI CC2531 dongle with this firmware https://github.com/virtualabs/cc2531-killerbee-fw for most things.

ep2277, 2024-04-04:
No, but most of the hardware given in this Repo is not available anymore. But you are right, the TI CC2531 is. We will try if we can do what we want to do with this hardware. Thank you for your quick answer. Thought I had to code an implementation for the HackRF which we are using in our LAB. Ordering the TI CC2531 for 60 bucks seems to be much easier 😄

gertrudem, 2024-04-05:
> I don't know of any available or anyone producing them. Is there something specific you're trying to do with the Apimote that other hardware won't support? I use the TI CC2531 dongle with this firmware https://github.com/virtualabs/cc2531-killerbee-fw for most things.

Just curious, what tools could you get working with the CC2531? I've been using it with the latest release of killerbee, but I've noticed that several tools do not seem to work. For example, zbdump did not work for any of the CC2531 firmware versions.
Did you bump into any issues like that?

ep2277, while the CC2531 is an option (might be the only option available), bear in mind that you might run into errors for some of the killerbee tools.

tc-26, 2024-04-05:
I use the library programmatically and write tools specifically for my use-cases for myself.

Here's an example to show how I do sniffing (output getting sent to wireshark in this example) and injecting packets.

```
import struct
import sys
import os
import threading
import subprocess
import time

from scapy.utils import PcapWriter
from scapy.layers.dot15d4 import Dot15d4
from scapy.config import conf
conf.dot15d4_protocol = "zigbee"

from killerbee.dev_bumblebee import *

########## Wireshark streaming and threading

def start_wireshark_capture(radio):
    ws = start_wireshark()
    pcap_writer = PcapWriter(ws.stdin)

    thread = threading.Thread(target=stream_out, args=(pcap_writer, radio))
    thread.start()

def stream_out(writer, radio):
    while True:
        try:
            packet = radio.pnext()
        except:
            packet = None

        if packet is not None:
            try:
                packet = Dot15d4(packet[0])
                writer.write(packet)
                writer.flush()
            except Exception as e:
                print(e)
                continue
        else:
            time.sleep(0.1)

def start_wireshark():
    spargs = dict(
        args=["wireshark", "-k", "-i", "-"],  # Read packets from stdin immediately
        stdin=subprocess.PIPE,
        stderr=open(os.devnull, "w"),
    )

    spargs["preexec_fn"] = os.setpgrp  # type: ignore

    ws = subprocess.Popen(**spargs)  # type: ignore

    return ws

##########

CC2531_USB_VEND_ID: int = 0x0451
BUMBLE_BEE_PROD_ID: int = 0x16A8

devices = usb.core.find(
    find_all=True,
    idVendor=CC2531_USB_VEND_ID,
    idProduct=BUMBLE_BEE_PROD_ID,
)

if devices is None:
    print("No devices found")
    sys.exit(0)

dev_rx = None
dev_tx = None

for dev in devices:
    if dev_tx is None:
        dev_tx = dev
    elif dev_rx is None:
        dev_rx = dev
    else:
        break

if dev_rx is None or dev_tx is None:
    print("Not enough devices found")
    sys.exit(0)

radio_rx = Bumblebee(dev_rx, None)
radio_tx = Bumblebee(dev_tx, None)

packet = b'taylor was here'

start_wireshark_capture(radio_rx)

while(True):
    radio_tx.inject(packet)
    time.sleep(1)

radio_rx.close()
radio_tx.close()
```

gertrudem, 2024-04-05:
Ah, I see. I will attempt at doing my own implementations.

Thank you